legubiao/quadruped_ros2_control #6: Rviz can't display go2

JeffXu1, 2024-10-14:
When I use command ros2 launch go2_description unitree_guide.launch.py, it display the following message:
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [20779]
[INFO] [robot_state_publisher-2]: process started with pid [20781]
[INFO] [ros2_control_node-3]: process started with pid [20783]
[INFO] [spawner-4]: process started with pid [20785]
[rviz2-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ros2_control_node-3] [INFO] [1728910210.139667234] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-3] [INFO] [1728910210.140282102] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-3] [WARN] [1728910210.140428610] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[robot_state_publisher-2] [INFO] [1728910210.141474063] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-2] [INFO] [1728910210.141576885] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-2] [INFO] [1728910210.141584995] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-2] [INFO] [1728910210.141589138] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-2] [INFO] [1728910210.141592601] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-2] [INFO] [1728910210.141595879] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-2] [INFO] [1728910210.141599233] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-2] [INFO] [1728910210.141602679] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-2] [INFO] [1728910210.141605945] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-2] [INFO] [1728910210.141609163] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-2] [INFO] [1728910210.141612391] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-2] [INFO] [1728910210.141615675] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-2] [INFO] [1728910210.141619010] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-2] [INFO] [1728910210.141622162] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-2] [INFO] [1728910210.141625484] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-2] [INFO] [1728910210.141628795] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-2] [INFO] [1728910210.141631956] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1728910210.141635199] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-2] [INFO] [1728910210.141638486] [robot_state_publisher]: got segment trunk
[ros2_control_node-3] [INFO] [1728910210.141775781] [controller_manager]: Received robot description file.
[ros2_control_node-3] [INFO] [1728910210.142109247] [resource_manager]: Loading hardware 'UnitreeSystem' 
[ros2_control_node-3] [INFO] [1728910210.146244600] [resource_manager]: Initialize hardware 'UnitreeSystem' 
[ros2_control_node-3] 1728910210.149395 [1] ros2_contr: selected interface "lo" is not multicast-capable: disabling multicast
[ros2_control_node-3] [INFO] [1728910210.254735905] [resource_manager]: Successful initialization of hardware 'UnitreeSystem'
[ros2_control_node-3] [INFO] [1728910210.255105833] [resource_manager]: 'configure' hardware 'UnitreeSystem' 
[ros2_control_node-3] [INFO] [1728910210.255121924] [resource_manager]: Successful 'configure' of hardware 'UnitreeSystem'
[ros2_control_node-3] [INFO] [1728910210.255151891] [resource_manager]: 'activate' hardware 'UnitreeSystem' 
[ros2_control_node-3] [INFO] [1728910210.255159725] [resource_manager]: Successful 'activate' of hardware 'UnitreeSystem'
[ros2_control_node-3] ros2_control_node: ./src/core/ddsc/src/dds_write.c:318: dds_writecdr_impl_common: Assertion `(wr->m_iox_pub == NULL) == (d->a.iox_chunk == NULL)' failed.
[rviz2-1] [INFO] [1728910210.370012717] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1728910210.370086139] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1728910210.383168232] [rviz2]: Stereo is NOT SUPPORTED
[spawner-4] [INFO] [1728910210.388325472] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[rviz2-1] Warning: Invalid frame ID "FL_hip" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "FL_thigh" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "FR_hip" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "FR_thigh" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "RL_hip" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "RL_thigh" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "RR_hip" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "RR_thigh" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "FL_hip" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "FL_thigh" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "FR_hip" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 93 in ./src/buffer_core.cpp
and there is no specified joint display on rviz
[image]

legubiao, 2024-10-15:
I think this error 
```
[ros2_control_node-3] ros2_control_node: ./src/core/ddsc/src/dds_write.c:318: dds_writecdr_impl_common: Assertion `(wr->m_iox_pub == NULL) == (d->a.iox_chunk == NULL)' failed.
```
Caused the controller manager died. This came from the unitree sdk2. Can you launch unitree mujoco simulation successfully?

hljmssjg, 2024-10-16:
@legubiao I also meet this problem. And I can launch unitree mujoco simulation successfully.

legubiao, 2024-10-16:
Are you using `sudo` to run the simulation, or simply `./unitree_mujoco`?

hljmssjg, 2024-10-16:
Follow the unitree mujoco document simply, that is , `./unitree_mojuco`

legubiao, 2024-10-16:
Could you provide your system and ros2 version? Also, some screenshots or log outputs.

hljmssjg, 2024-10-16:
I found that although the error I encountered in RViz looks similar to the issue mentioned, my problem was actually different. I already resolved it by configuring the SDK library files.  :)
By the way, is your Unitree SDK also installed in the /opt/unitree_robotics directory?

legubiao, 2024-10-16:
I just used `sudo make install` and the path seems are ` /usr/local/include/unitree` and ` /usr/local/include/dds`. You are welcomed to open another issue for your error.

closed 2024-10-28 by legubiao