legubiao/quadruped_ros2_control #7: Building error on Ubuntu 20.04 with ROS2 Galastic

xander-2077, 2024-10-16:
This is a fun project! But has anyone tried to build this project on Ubuntu 20.04 with ROS2 Galastic?

legubiao, 2024-10-17:
I don't recommend using ROS2 Galastic with unitree_sdk2, since the default DDS for Galastic is CyloneDDS and it has some problem with unitree_sdk2. Now I'm testing the project under ros2 Foxy, later I 'll create a new branch for foxy.

xander-2077, 2024-10-17:
Thanks for your rapid response! I have built unitree_ros2 before. It seems that the unitree_ros2 replaces the default DDS with an older version. Looking forward to your new branch!

legubiao, 2024-10-17:
I just check the project under Ubuntu 20.04, seems that ros2-control framework has a quiet large gap between foxy and humble. The project code needs many modifications to fit foxy. Right now, I'm still focusing on the development of new controllers, so I don't have enough time for foxy branch recently.  If you have interest, you can try to fork your own foxy version.

xander-2077, 2024-10-17:
Get it! It seems that the ubuntu 22.04 is the better choice. Thank you!

closed 2024-10-17 by xander-2077