legubiao/quadruped_ros2_control #9: Problems when compiling the project

LanceXu2001, 2024-10-26:
Thanks for your sharing of the excellent project. But when I use this command to compile the pd_controller

colcon build --packages-up-to leg_pd_controller ,

the error shows 

/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:218:12: error: invalid new-expression of abstract class type ‘leg_pd_controller::LegPdController’
  218 |     return new C;
      |            ^~~~~

How can I fix it?  Would this problems arise from the wrong version of controller_manager? I change 

find_package(controller_manager 3.0.0)  -> find_package(controller_manager 2.43.1)

in the CMakelist for compile.

legubiao, 2024-10-26:
This error is because the ROS2-Control version is mismatch with the controller. The reason I add version check for `controller_manager` is that their implementation for `controller_interface::ChainableControllerInterface` is different between ROS2 Humble (2.0) and ROS2 Jazzy (4.0). You can check the `official ROS2_Control demo 12 (passthrough_controller)` as an example and modified the  leg_pd_controller.

legubiao, 2024-10-26:
This change disabled version check, so you are using the code for ROS2 Jazzy but actually you are using ROS2 Humble:
`find_package(controller_manager 3.0.0) -> find_package(controller_manager 2.43.1)`

LanceXu2001, 2024-10-27:
The problem is solved. THX!

LanceXu2001, 2024-10-27:
I meet another error when running the project. when I use  `ros2 launch unitree_guide_controller gazebo.launch.py` and the terminal shows ```
[ruby $(which ign) gazebo-3] [INFO] [1730031533.123817398] [controller_manager]: Loading controller 'leg_pd_controller'
[ruby $(which ign) gazebo-3] Exception thrown during init stage with message: parameter 'state_interfaces' has already been declared 
[ruby $(which ign) gazebo-3] [ERROR] [1730031533.128083024] [controller_manager]: Could not initialize the controller named 'leg_pd_controller'
[spawner-6] [FATAL] [1730031533.130178509] [spawner_leg_pd_controller]: Failed loading controller leg_pd_controller
[ruby $(which ign) gazebo-3] [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [St6vectorIdSaIdEE], which doesn't have `operator>>`. Component will not be deserialized.
[ERROR] [spawner-6]: process has died [pid 147214, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner leg_pd_controller --controller-manager /controller_manager --ros-args']. while the gazebo and rviz run normally.
```

LanceXu2001, 2024-10-27:
I meet another error when launch the program. When I use `ros2 launch unitree_guide_controller gazebo.launch.py` the terminal shows ```
[ruby $(which ign) gazebo-3] [INFO] [1730031533.123817398] [controller_manager]: Loading controller 'leg_pd_controller'
[ruby $(which ign) gazebo-3] Exception thrown during init stage with message: parameter 'state_interfaces' has already been declared 
[ruby $(which ign) gazebo-3] [ERROR] [1730031533.128083024] [controller_manager]: Could not initialize the controller named 'leg_pd_controller'
[spawner-6] [FATAL] [1730031533.130178509] [spawner_leg_pd_controller]: Failed loading controller leg_pd_controller
[ruby $(which ign) gazebo-3] [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [St6vectorIdSaIdEE], which doesn't have `operator>>`. Component will not be deserialized.
[ERROR] [spawner-6]: process has died [pid 147214, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner leg_pd_controller --controller-manager /controller_manager --ros-args'].

``` while gazebo and rviz work normally.

LanceXu2001, 2024-10-27:
I meet another error when I launch the program. After I use `ros2 launch unitree_guide_controller gazebo.launch.py `, the terminal shows 

[ruby $(which ign) gazebo-3] [INFO] [1730031533.123817398] [controller_manager]: Loading controller 'leg_pd_controller'
[ruby $(which ign) gazebo-3] Exception thrown during init stage with message: parameter 'state_interfaces' has already been declared 
[ruby $(which ign) gazebo-3] [ERROR] [1730031533.128083024] [controller_manager]: Could not initialize the controller named 'leg_pd_controller'
[spawner-6] [FATAL] [1730031533.130178509] [spawner_leg_pd_controller]: Failed loading controller leg_pd_controller
[ruby $(which ign) gazebo-3] [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [St6vectorIdSaIdEE], which doesn't have `operator>>`. Component will not be deserialized.
[ERROR] [spawner-6]: process has died [pid 147214, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner leg_pd_controller --controller-manager /controller_manager --ros-args'].

while the gazebo and rviz can launch normally.

legubiao, 2024-10-27:
I haven't test Gazebo simulation under Ubuntu 22.04(ROS2 Humble). I only test Gazebo Classic simulation, you can try that one.

LanceXu2001, 2024-10-27:
I have tried just now, and it shows the same mistake.

legubiao, 2024-10-27:
I think this error 
`Exception thrown during init stage with message: parameter 'state_interfaces' has already been declared` 
is the key issue. You may have some residual programs that you haven't closed. How about reboot system and have another try?

LanceXu2001, 2024-10-27:
Unfortunately, the same issue happens when I reboot the system. I found that `Exception thrown during init stage with message: parameter 'state_interfaces' has already been declared` shows when I comment the `joints:....` of leg_pd_controller in go2_description/config/gazebo.yaml and if I comment out `joints:....` the terminal shows  

[gzserver-3] [INFO] [1730037199.946045557] [controller_manager]: Loading controller 'leg_pd_controller'
[gzserver-3] Exception thrown during init stage with message: parameter 'joints' has already been declared 
[gzserver-3] [ERROR] [1730037199.954341245] [controller_manager]: Could not initialize the controller named 'leg_pd_controller'
[spawner-6] [FATAL] [1730037199.957454099] [spawner_leg_pd_controller]: Failed loading controller leg_pd_controller
[ERROR] [spawner-6]: process has died [pid 14663, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner leg_pd_controller --controller-manager /controller_manager --ros-args'].

legubiao, 2024-10-27:
OK, later I will use a fresh WSL to check it.

legubiao, 2024-10-28:
I just fixed some bugs and test pass on WSL. You can have another try.

LanceXu2001, 2024-10-28:
The program runs successfully. Thanks for your help!🎉

closed 2024-10-28 by legubiao