legubiao/quadruped_ros2_control #10: Building error ocs2_quadruped_controller

af-yi, 2024-10-30:
你好，我根据 https://github.com/legubiao/quadruped_ros2_control/tree/main/controllers/ocs2_quadruped_controller , 编译ocs2_quadruped_controller时，出现了下面的编译问题，似乎是控制器get_node()获取的类型和leggedRobotVisualizer接收的类型不匹配。不知你有没有遇见过。
cd ~/ros2_ws colcon build --packages-up-to ocs2_quadruped_controller
[image]
 
[image]

legubiao, 2024-10-30:
leggedRobotVisualizer我增加了一个构造函数是传入LifecycleNode的，你可以检查下ocs2_ros2的代码里有没有这一部分

af-yi, 2024-10-30:
> 我注意到截图中的ocs2_ros2确实不是最新版本，拉取下代码然后重新编译应该能解决你的问题

感谢，
解决了，是ocs2不是最新版本。

closed 2024-10-30 by af-yi

linClubs, 2025-01-20:
@af-yi 请问您是拉取的那个版本代码解决此问题，方面给个链接么，感谢

legubiao, 2025-01-20:
> [@af-yi](https://github.com/af-yi) 请问您是拉取的那个版本代码解决此问题，方面给个链接么，感谢

在我主页可以找到我fork的版本。这个仓库中ocs2 controller文件夹下的readme有对应的说明