legubiao/quadruped_ros2_control #11: rviz和gazebo显示不对，也不能控制

JeffXu1, 2024-10-31:
按照1.2 Gazebo Classic Simulator (ROS2 Humble)这个步骤进行测试，系统是Ubuntu22 + ROS Humble Gazebo Classic
但是显示的不对,使用ros2 run keyboard_input keyboard_input命令之后也不能控制，已经在root模式下
[image]
[image]
[image]

legubiao, 2024-10-31:
有按照keyboard input的readme尝试控制吗？此外，我看报错路径有个xxgo_description，似乎不是我的项目里的内容？

legubiao, 2024-10-31:
显示倒是看起来没问题，rviz和gazebo中的是同步的。得让机器人站起来才是readme中的效果

legubiao, 2024-11-03:
如果没有更多的反馈，我会在一天后关闭这个议题

closed 2024-11-04 by legubiao