legubiao/quadruped_ros2_control #12: 在运行rl controller的时候报错

hljmssjg, 2024-11-07:
[image]
报了一个symbol lookup error：

[ros2_control_node-3] /opt/ros/humble/lib/controller_manager/ros2_control_node: symbol lookup error: /home/jiangeng/ros2_ws/install/rl_quadruped_controller/lib/rl_quadruped_controller/librl_quadruped_controller.so: undefined symbol: _ZN3c108ListType3getENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_4Type24SingletonOrSharedTypePtrIS7_EE
[ERROR] [ros2_control_node-3]: process has died [pid 459726, exit code 127, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /home/jiangeng/ros2_ws/install/go2_description/share/go2_description/config/robot_control.yaml -r ~/robot_description:=/robot_description'].
[spawner-6] [WARN] [1730975106.607939379] [spawner_rl_quadruped_controller]: Failed getting a result from calling /controller_manager/switch_controller in 0.0. (Attempt 1 of 3.)
[spawner-6] [WARN] [1730975116.608949833] [spawner_rl_quadruped_controller]: Failed getting a result from calling /controller_manager/switch_controller in 0.0. (Attempt 2 of 3.)
[spawner-6] [WARN] [1730975126.609651696] [spawner_rl_quadruped_controller]: Failed getting a result from calling /controller_manager/switch_controller in 0.0. (Attempt 3 of 3.)
[spawner-6] Traceback (most recent call last):
[spawner-6]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
[spawner-6]     sys.exit(load_entry_point('controller-manager==2.43.1', 'console_scripts', 'spawner')())
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 303, in main
[spawner-6]     ret = switch_controllers(
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/controller_manager_services.py", line 207, in switch_controllers
[spawner-6]     return service_caller(
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/controller_manager_services.py", line 90, in service_caller
[spawner-6]     raise RuntimeError(
[spawner-6] RuntimeError: Could not successfully call service /controller_manager/switch_controller after 3 attempts.
[ERROR] [spawner-6]: process has died [pid 459848, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner rl_quadruped_controller --controller-manager /controller_manager --ros-args'].

legubiao, 2024-11-07:
看起来像是libtorch没配好，检查下是不是c++11以上的版本，还有LD_LIBRARY_PATH有没有正确配置。或者系统内可能有其他的torch影响到

hljmssjg, 2024-11-08:
重安了下libtorch+重新build下项目，问题已解决。

closed 2024-11-08 by hljmssjg