legubiao/quadruped_ros2_control #13: 使用rl controller和keyboard controller控制a1从fixed stand to rl时候出现Error，控制go2无任何反应

IsabeldeCastilla, 2024-11-07:
使用 rl controller + keyboard controller 控制a1：
问题：只能按两次“键2”,使a1变成fixed down，再变成fixed stand，再按“键3”就会出现switched to rl然后出现error
command:
ros2 launch rl_quadruped_controller gazebo_classic.launch.py pkg_description:=a1_description
ros2 run keyboard_input keyboard_input

Error:
[gzserver-3] [INFO] [1731021064.569754131] [rl_quadruped_controller]: Switched from fixed down to fixed stand
[gzserver-3] [INFO] [1731021068.503151901] [rl_quadruped_controller]: Switched from fixed stand to rl
[gzserver-3] corrupted size vs. prev_size
[ERROR] [gzserver-3]: process has died [pid 66993, exit code -6, cmd 'gzserver -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].

使用 rl controller + keyboard controller 控制go2：
使用 “键2” 无任何反应，terminal无switch to xxx 的 提示，gazebo画面也没有任何变化

legubiao, 2024-11-08:
gazebo classic是这样的，切到rl的时候会直接死掉，暂时没有解决方法。你可以试试用mujoco和modern gazebo，这两个是测试过可以跑rl的。

IsabeldeCastilla, 2024-11-11:
请问是在Ubuntu 24 + ros2 Jazzy上测试的吗？似乎使用Ubuntu 22.04 + ros2 humble + gazebo fortress的时候，gz-ros2-control（我自己build from source然后改了libarary名字）这个plugin有问题，控制没有任何反应：
[ruby $(which ign) gazebo-3] [Err] [SystemLoader.cc:125] Failed to load system plugin [gz_ros2_control::GazeboSimROS2ControlPlugin] : could not instantiate from library [gz_ros2_control-system] from path [......./ign_ros2_control_ws/install/ign_ros2_control/lib/libgz_ros2_control-system.so].

而使用mujoco的话，我出现的情况是按站立按键的话，go2会直接站起来向后翻倒在地。

IsabeldeCastilla, 2024-11-11:
> 请问是在Ubuntu 24 + ros2 Jazzy上测试的吗？似乎使用Ubuntu 22.04 + ros2 humble + gazebo fortress的时候，gz-ros2-control（我自己build from source然后改了libarary名字）这个plugin有问题，控制没有任何反应： [ruby $(which ign) gazebo-3] [Err] [...]

legubiao, 2024-11-12:
你用的mujoco仿真里面这个是云深处的x30机器人，而不是宇树的go2，得改下unitree mujoco的配置文件里面的机器人型号

legubiao, 2024-11-12:
modern gazebo是在ROS2 jazzy下测试的，我没有测试过Humble下编译安装的插件

IsabeldeCastilla, 2024-11-18:
> 你用的mujoco仿真里面这个是云深处的x30机器人，而不是宇树的go2，得改下unitree mujoco的配置文件里面的机器人型号

换成go2之后已解决。十分感谢！