legubiao/quadruped_ros2_control #19: colcon build --packages-up-to unitree_guide_controller go2_description keyboard_input --cmake-args -Wno-dev

yifengzhong-cat, 2025-02-18:


legubiao, 2025-02-19:
Could you please provide more details about your problems?

yifengzhong-cat, 2025-02-19:
我的意思是，我要在后面加-Wno-dev 才能成功，但是ros2 launch unitree_guide_controller mujoco.launch.py mujoco 启动不了呢

legubiao, 2025-02-19:
launch并不会启动mujoco，mujoco需要另外启动，参考readme中的说明。至于前面的编译问题，提供下你的系统环境版本信息？

JINGERGER, 2025-02-19:
[image]
@legubiao ,我根据教程能跑，但是感觉键盘延时很大

legubiao, 2025-02-19:
感觉可能不是延时的问题而是键盘输入节点的机制，可以根据需要直接修改键盘输入节点的代码。或者，尝试使用手柄进行输入

legubiao, 2025-02-19:
> 我的意思是，我要在后面加-Wno-dev 才能成功，但是ros2 launch unitree_guide_controller mujoco.launch.py mujoco 启动不了呢

去掉了因为ros2 control更新弃用导致没法编译通过的注解，刚刚测了下在22和24下都是顺利编译通过的，不太确定是不是和你的这个问题相关，可以再尝试一下

JINGERGER, 2025-02-20:
@legubiao, 手柄很丝滑

> 感觉可能不是延时的问题而是键盘输入节点的机制，可以根据需要直接修改键盘输入节点的代码。或者，尝试使用手柄进行输入

closed 2025-03-15 by legubiao