legubiao/quadruped_ros2_control #23: 遇到了一堆奇怪的编译问题

jackhou66, 2025-03-22:
C++ 17 compile ros2 foxy  

/quadruped_ros2_control/libraries/controller_common/src/FSM/BaseFixedStand.cpp: In member function ‘virtual void BaseFixedStand::enter()’:
/home/jack/Desktop/sim/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/src/FSM/BaseFixedStand.cpp:26:83: error: ‘class hardware_interface::LoanedStateInterface’ has no member named ‘get_optional’
   26 |         start_pos_[i] = ctrl_interfaces_.joint_position_state_interface_[i].get().get_optional().value();
      |                                                                                   ^~~~~~~~~~~~
/home/jack/Desktop/sim/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/src/FSM/BaseFixedStand.cpp:30:106: error: no match for ‘operator=’ (operand types are ‘const std::_Swallow_assign’ and ‘void’)
   30 |         std::ignore = ctrl_interfaces_.joint_position_command_interface_[i].get().set_value(start_pos_[i]);
      |                                                                                                          ^
/quadruped_ros2_control/libraries/controller_common/src/FSM/BaseFixedStand.cpp:30:106:   required from here
/usr/include/c++/11/tuple:1779:7: error: forming reference to void
/usr/include/c++/11/tuple:1775:10: note: candidate: ‘constexpr std::_Swallow_assign& std::_Swallow_assign::operator=(const std::_Swallow_assign&)’
 1775 |   struct _Swallow_assign

legubiao, 2025-03-23:
ros2 foxy下的ros2 control API和humble之后的差别挺大的，要改的东西很多，没有做专门的适配。

mrleejohnson, 2025-03-27:
我用的是humble也会出现这样的问题/home/leejohnson/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/src/FSM/BaseFixedStand.cpp: In member function ‘virtual void BaseFixedStand::run(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/leejohnson/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/src/FSM/BaseFixedStand.cpp:46:65: error: no match for ‘operator=’ (operand types are ‘const std::_Swallow_assign’ and ‘void’)
   46 |             phase * target_pos_[i] + (1 - phase) * start_pos_[i]);
      |                                                                 ^
In file included from /usr/include/c++/11/functional:54,
                 from /opt/ros/humble/include/hardware_interface/loaned_command_interface.hpp:18,
                 from /home/leejohnson/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/include/controller_common/CtrlInterfaces.h:9,
                 from /home/leejohnson/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/include/controller_common/FSM/FSMState.h:11,
                 from /home/leejohnson/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/include/controller_common/FSM/BaseFixedStand.h:6,
                 from /home/leejohnson/ros2_ws/src/quadruped_ros2_control/libraries/controller_common/src/FSM/BaseFixedStand.cpp:5:
/usr/include/c++/11/tuple:1779:7: note: candidate: ‘template<class _Tp> constexpr const std::_Swallow_assign& std::_Swallow_assign::operator=(const _Tp&) const’
 1779 |       operator=(const _Tp&) const
      |       ^~~~~~~~
/usr/include/c++/11/tuple:1779:7: note:   template argument deduction/substitution failed:
/usr/include/c++/11/tuple:1775:10: note: candidate: ‘constexpr std::_Swallow_assign& std::_Swallow_assign::operator=(const std::_Swallow_assign&)’
 1775 |   struct _Swallow_assign
      |          ^~~~~~~~~~~~~~~
/usr/include/c++/11/tuple:1775:10: note:   no known conversion for argument 1 from ‘void’ to ‘const std::_Swallow_assign&’
/usr/include/c++/11/tuple:1775:10: note: candidate: ‘constexpr std::_Swallow_assign& std::_Swallow_assign::operator=(std::_Swallow_assign&&)’
/usr/include/c++/11/tuple:1775:10: note:   no known conversion for argument 1 from ‘void’ to ‘std::_Swallow_assign&&’
gmake[2]: *** [CMakeFiles/controller_common.dir/build.make:90：CMakeFiles/controller_common.dir/src/FSM/BaseFixedStand.cpp.o] 错误 1
gmake[1]: *** [CMakeFiles/Makefile2:137：CMakeFiles/controller_common.dir/all] 错误 2
gmake: *** [Makefile:146：all] 错误 2
---
Failed   <<< controller_common [0.79s, exited with code 2]

legubiao, 2025-03-28:
重新检查了下，发现是jazzy的api相比humble发生了新的变化。现在我创建了humble的分支来处理这些情况了

closed 2025-04-01 by legubiao