legubiao/quadruped_ros2_control #26: 运行ros2 launch rl_quadruped_controller mujoco.launch.py pkg_description:=go2_description无法控制

KUsaka-xu, 2025-04-01:
[image]
在跑RL控制时出现Failed loading controller rl_quadruped_controller

[INFO] [launch]: All log files can be found below /home/saka/.ros/log/2025-04-01-23-28-51-342179-saka-System-Product-Name-4498
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [4501]
[INFO] [robot_state_publisher-2]: process started with pid [4502]
[INFO] [ros2_control_node-3]: process started with pid [4503]
[INFO] [spawner-4]: process started with pid [4504]
[robot_state_publisher-2] [INFO] [1743517731.438130150] [robot_state_publisher]: Robot initialized
[ros2_control_node-3] [INFO] [1743517731.452902617] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-3] [INFO] [1743517731.454575215] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-3] [INFO] [1743517731.454583873] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-3] [WARN] [1743517731.454615728] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-3] [INFO] [1743517731.463118894] [controller_manager]: Received robot description from topic.
[ros2_control_node-3] [INFO] [1743517731.466038367] [controller_manager.resource_manager]: Loading hardware 'UnitreeSystem' 
[ros2_control_node-3] [INFO] [1743517731.468567260] [controller_manager.resource_manager]: Loaded hardware 'UnitreeSystem' from plugin 'hardware_unitree_mujoco/HardwareUnitree'
[ros2_control_node-3] [INFO] [1743517731.468594201] [controller_manager.resource_manager]: Initialize hardware 'UnitreeSystem' 
[ros2_control_node-3] [INFO] [1743517731.468786932] [controller_manager.resource_manager.hardware_component.system.UnitreeSystem]:  network_interface: lo, domain: 1
[ros2_control_node-3] 1743517731.469706 [1] ros2_contr: selected interface "lo" is not multicast-capable: disabling multicast
[ros2_control_node-3] [INFO] [1743517731.572235337] [controller_manager.resource_manager]: Successful initialization of hardware 'UnitreeSystem'
[ros2_control_node-3] [INFO] [1743517731.572661313] [resource_manager]: 'configure' hardware 'UnitreeSystem' 
[ros2_control_node-3] [INFO] [1743517731.572666300] [resource_manager]: Successful 'configure' of hardware 'UnitreeSystem'
[ros2_control_node-3] [INFO] [1743517731.572674894] [resource_manager]: 'activate' hardware 'UnitreeSystem' 
[ros2_control_node-3] [INFO] [1743517731.572677222] [resource_manager]: Successful 'activate' of hardware 'UnitreeSystem'
[ros2_control_node-3] [INFO] [1743517731.572682248] [controller_manager]: Resource Manager has been successfully initialized. Starting Controller Manager services...
[ros2_control_node-3] [INFO] [1743517731.620000675] [controller_manager]: Loading controller : 'joint_state_broadcaster' of type 'joint_state_broadcaster/JointStateBroadcaster'
[ros2_control_node-3] [INFO] [1743517731.620020578] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2_control_node-3] [INFO] [1743517731.621214017] [controller_manager]: Controller 'joint_state_broadcaster' node arguments: --ros-args --params-file /home/saka/ros2_ws/install/go2_description/share/go2_description/config/robot_control.yaml 
[spawner-4] [INFO] [1743517731.629605687] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1743517731.629897491] [controller_manager]: Configuring controller: 'joint_state_broadcaster'
[ros2_control_node-3] [INFO] [1743517731.629948501] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ros2_control_node-3] [INFO] [1743517731.634418269] [controller_manager]: Activating controllers: [ joint_state_broadcaster ]
[spawner-4] [INFO] [1743517731.636318878] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[INFO] [spawner-4]: process has finished cleanly [pid 4504]
[INFO] [spawner-5]: process started with pid [4602]
[ros2_control_node-3] [INFO] [1743517731.951930442] [controller_manager]: Loading controller : 'imu_sensor_broadcaster' of type 'imu_sensor_broadcaster/IMUSensorBroadcaster'
[ros2_control_node-3] [INFO] [1743517731.951950847] [controller_manager]: Loading controller 'imu_sensor_broadcaster'
[ros2_control_node-3] [INFO] [1743517731.952293001] [controller_manager]: Controller 'imu_sensor_broadcaster' node arguments: --ros-args --params-file /home/saka/ros2_ws/install/go2_description/share/go2_description/config/robot_control.yaml 
[spawner-5] [INFO] [1743517731.960638115] [spawner_imu_sensor_broadcaster]: Loaded imu_sensor_broadcaster
[ros2_control_node-3] [INFO] [1743517731.960934033] [controller_manager]: Configuring controller: 'imu_sensor_broadcaster'
[ros2_control_node-3] [INFO] [1743517731.962243905] [controller_manager]: Activating controllers: [ imu_sensor_broadcaster ]
[spawner-5] [INFO] [1743517731.964225015] [spawner_imu_sensor_broadcaster]: Configured and activated imu_sensor_broadcaster
[INFO] [spawner-5]: process has finished cleanly [pid 4602]
[INFO] [spawner-6]: process started with pid [4618]
[rviz2-1] [INFO] [1743517732.150546779] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1743517732.150590902] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ros2_control_node-3] [INFO] [1743517732.276519593] [controller_manager]: Loading controller : 'rl_quadruped_controller' of type 'rl_quadruped_controller/LeggedGymController'
[ros2_control_node-3] [INFO] [1743517732.276538683] [controller_manager]: Loading controller 'rl_quadruped_controller'
[ros2_control_node-3] [ERROR] [1743517732.276905719] [controller_manager]: Caught exception of type : N9pluginlib20LibraryLoadExceptionE while loading the controller 'rl_quadruped_controller' of plugin type 'rl_quadruped_controller/LeggedGymController':
[ros2_control_node-3] Failed to load library /home/saka/ros2_ws/install/rl_quadruped_controller/lib/rl_quadruped_controller/librl_quadruped_controller.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libtorch_cuda.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
[spawner-6] [FATAL] [1743517732.282497212] [spawner_rl_quadruped_controller]: Failed loading controller rl_quadruped_controller
[rviz2-1] [INFO] [1743517732.295744719] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [spawner-6]: process has died [pid 4618, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner rl_quadruped_controller --controller-manager /controller_manager --ros-args'].

legubiao, 2025-04-01:
Error string: Could not load library dlopen error: libtorch_cuda.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
可以往libtorch配置的思路去排查下，我估计是LD_LIBRARY_PATH没有配置好，虽然能编译通过但是没法正常运行

KUsaka-xu, 2025-04-02:
我按照cuda版本重新装了libtorch，问题解决了，感谢大佬的帮助

rm -rf ~/CLionProjects/libtorch
cd ~/CLionProjects/
wget https://download.pytorch.org/libtorch/cu121/libtorch-cxx11-abi-shared-with-deps-2.5.0%2Bcu121.zip -O libtorch.zip
unzip libtorch.zip
rm libtorch.zip
echo 'export Torch_DIR=~/CLionProjects/libtorch' >> ~/.bashrc
echo 'export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/CLionProjects/libtorch/lib' >> ~/.bashrc
source ~/.bashrc
然后重新build清理cache
cd ~/ros2_ws
colcon build --packages-select rl_quadruped_controller --symlink-install --cmake-clean-cache

closed 2025-04-02 by KUsaka-xu