legubiao/quadruped_ros2_control #29: unitree_guide_controller mujoco.launch.py仿真报错

diaoweiqing, 2025-04-17:
请问RMW—implementation是怎么设置的，在unitree_guide_controller mujoco.launch.py中，显示[ERROR] [ros2_control_node-3]: process has died [pid 29672, exit code -6, cmd '/opt/ros/jazzy/lib/controller_manager/ros2_control_node --ros-args --params-file /home/dwq/ros2_ws/install/go2_description/share/go2_description/config/robot_control.yaml -r ~/robot_description:=/robot_description'].
[rviz2-1] [INFO] [1744884015.476549450] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1744884015.476624558] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-1] [INFO] [1744884015.522218501] [rviz2]: Stereo is NOT SUPPORTED
[spawner-4] [WARN] [1744884017.992580255] [spawner_joint_state_broadcaster]: Controller already loaded, skipping load_controller
[spawner-4] [ERROR] [1744884018.076907097] [spawner_joint_state_broadcaster]: Failed to configure controller
[ERROR] [spawner-4]: process has died [pid 29673, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[INFO] [spawner-5]: process started with pid [29772]
[spawner-5] [WARN] [1744884019.015250433] [spawner_imu_sensor_broadcaster]: Controller already loaded, skipping load_controller
[spawner-5] [ERROR] [1744884019.024774849] [spawner_imu_sensor_broadcaster]: Failed to configure controller
[ERROR] [spawner-5]: process has died [pid 29772, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner imu_sensor_broadcaster --controller-manager /controller_manager --ros-args'].
[INFO] [spawner-6]: process started with pid [29787]
[spawner-6] [WARN] [1744884019.656988759] [spawner_unitree_guide_controller]: Controller already loaded, skipping load_controller
[spawner-6] [ERROR] [1744884019.662189174] [spawner_unitree_guide_controller]: Failed to configure controller
[ERROR] [spawner-6]: process has died [pid 29787, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner unitree_guide_controller --controller-manager /controller_manager --ros-args'].

legubiao, 2025-04-19:
你这里的报错的内容看起来是因为之前的控制器的进程没有清理干净的样子？`Controller already loaded, skipping load_controller`

diaoweiqing, 2025-04-22:
博主您好，在您的ocsquadrupedcontroller中，编译出现了如下问题

diaoweiqing, 2025-04-22:
博主您好，在您的OCS2quadrupedcontroller编译中，出现了如下的问题，想问一下您有遇到过吗

legubiao, 2025-04-22:
看不到你发的问题的内容

diaoweiqing, 2025-04-22:
编译ocs2_quadruped_controller时报错

[image]

legubiao, 2025-04-22:
没有遇到过类似的情况。我有注意到你是在virtualbox里编译的，可能有这里的影响，或许你可以考虑采用原系统或者用distrobox开的环境

diaoweiqing, 2025-04-23:
博主你好，在编译ocs2_quadruped_controller包下的ocs2_sphere_approximation出现了如下报错，请问博主是否遇见过

[image]

legubiao, 2025-04-23:
hpp fcl的问题。可能是因为你的pinocchio的安装不是按照文档里写的方法安装的

diaoweiqing, 2025-04-23:
请问博主的hpp-fcl是怎么安装的，是手动安装还是sudo apt-get install ros-jazzy-hpp-fcl

legubiao, 2025-04-23:
不是通过ros安装的。在ocs2 ros2的readme中有关于pinnocchio安装的Warm Reminder

closed 2025-05-22 by legubiao