legubiao/quadruped_ros2_control #30: ubuntu22.04 humble  ocs2 mujoco 一按下2 机器人就乱飞然后进入安全模式

TheLionel, 2025-04-18:
不知道是不是代码是24.04的缘故 中间出现了不少代码上的问题 比如std::ignore,get_optional.value()等 一一改正过后 终于能运行 但是表现出来的效果很奇怪，不能正在走

TheLionel, 2025-04-18:
[image]

legubiao, 2025-04-18:
切换到humble分支看看呢？

TheLionel, 2025-04-18:
> 切换到humble分支看看呢？

我去  我没注意看！多谢作者提醒

TheLionel, 2025-04-18:
> 切换到humble分支看看呢？

谢谢作者 正常跑了 不过切换过去也发现了不少编译上的错误 hardware_interface中获取值的接口get_optional.value() 还有参数声明重复 我是humble 22.04 改了就好了

[image]

legubiao, 2025-04-18:
暂时没适配humble下的gazebo呢

TheLionel, 2025-04-19:
> 暂时没适配humble下的gazebo呢

好的 谢谢！ 后续有计划嘛 我先阅读下代码学习学习

legubiao, 2025-04-19:
尝试适配了下，humble下可以用gazebo harmonic配合unitree guide或者rl的控制器使用，ocs2的控制器会碰上很奇怪的bug。只能说还是在jazzy下更舒心一点

TheLionel, 2025-04-19:
> 尝试适配了下，humble下可以用gazebo harmonic配合unitree guide或者rl的控制器使用，ocs2的控制器会碰上很奇怪的bug。只能说还是在jazzy下更舒心一点

好的 感谢感谢！

closed 2025-04-19 by TheLionel