legubiao/quadruped_ros2_control #31: 1744991603.030922 [1] unitree_mu: selected interface "lo" is not multicast-capable: disabling multicast free(): invalid pointer 已中止 (核心已转储)

lkn01, 2025-04-18:
只想使用mujoco的部分，根据readme配置，一开始没留意warn里面的东西在unitree_mujoco里面运行./unitree_mujoco还能出现mujoco界面中的go2结果按照warn里的说法按照第二条配置了
Follow the guide in [unitree_ros2](https://github.com/unitreerobotics/unitree_ros2) to configure the ROS2 RMW by compiling cyclone dds.

然后运行./unitree_mujoco说"lo" 这个有问题？
user@user:~/ros2_mujoco_test/quadruped_ros2_control/src/unitree_mujoco/simulate/build$ ./unitree_mujoco
MuJoCo version 3.3.2
Mujoco data is prepared
1744991603.030922 [1] unitree_mu: selected interface "lo" is not multicast-capable: disabling multicast
free(): invalid pointer
已中止 (核心已转储)
 
一时半会没搞懂

legubiao, 2025-04-18:
这个警报可以忽略`unitree_mu: selected interface "lo" is not multicast-capable: disabling multicast`
如果你使用的是ros2 humble，那你其实默认用的就不是cyclonedds rmw，不需要额外的配置

lkn01, 2025-04-18:
> 这个警报可以忽略`unitree_mu: selected interface "lo" is not multicast-capable: disabling multicast` 如果你使用的是ros2 humble，那你其实默认用的就不是cyclonedds rmw，不需要额外的配置

但是现在运行mujoco的会闪退，有啥办法能解决吗，一开始还不闪退的，配置了一些东西后运行就闪退了。
环境确实是ubuntu22.04 ros2 humble

legubiao, 2025-04-18:
想办法去除cyclonedds的影响

lkn01, 2025-04-19:
> 想办法去除cyclonedds的影响

加了sudo启动，可以了

closed 2025-04-19 by lkn01