legubiao/quadruped_ros2_control #32: 运行ros2控制器报错了

lkn01, 2025-04-19:
环境：ubuntu22.04,  ros2 humble ， 运行 ros2 mujoco b2的部分

运行ros2控制器报错了
user@user:~/ros2_mujoco_test/quadruped_ros2_control$ ros2 launch unitree_guide_controller mujoco.launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2025-04-19-14-40-50-381035-user-36123
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [36124]
[INFO] [robot_state_publisher-2]: process started with pid [36126]
[INFO] [ros2_control_node-3]: process started with pid [36128]
[INFO] [spawner-4]: process started with pid [36130]
[robot_state_publisher-2] 1745044850.465315 [0] robot_stat: selected interface "lo" is not multicast-capable: disabling multicast
[ros2_control_node-3] 1745044850.467578 [0] ros2_contr: selected interface "lo" is not multicast-capable: disabling multicast
[ros2_control_node-3] [INFO] [1745044850.476136327] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-3] [INFO] [1745044850.476435728] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-3] [INFO] [1745044850.476446779] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-3] [WARN] [1745044850.476509088] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[robot_state_publisher-2] [INFO] [1745044850.483970437] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-2] [INFO] [1745044850.484016155] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-2] [INFO] [1745044850.484021638] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-2] [INFO] [1745044850.484025494] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-2] [INFO] [1745044850.484029255] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-2] [INFO] [1745044850.484032812] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-2] [INFO] [1745044850.484036287] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-2] [INFO] [1745044850.484039914] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-2] [INFO] [1745044850.484043590] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-2] [INFO] [1745044850.484047182] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-2] [INFO] [1745044850.484050623] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-2] [INFO] [1745044850.484054126] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-2] [INFO] [1745044850.484057552] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-2] [INFO] [1745044850.484060993] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-2] [INFO] [1745044850.484064376] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-2] [INFO] [1745044850.484067831] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-2] [INFO] [1745044850.484071277] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1745044850.484074702] [robot_state_publisher]: got segment front_camera
[robot_state_publisher-2] [INFO] [1745044850.484078193] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-2] [INFO] [1745044850.484081651] [robot_state_publisher]: got segment lidar
[robot_state_publisher-2] [INFO] [1745044850.484085037] [robot_state_publisher]: got segment trunk
[ros2_control_node-3] [INFO] [1745044850.484185180] [controller_manager]: Received robot description file.
[ros2_control_node-3] [INFO] [1745044850.484520653] [resource_manager]: Loading hardware 'UnitreeSystem' 
[ros2_control_node-3] [ERROR] [1745044850.484965722] [controller_manager]: The published robot description file (urdf) seems not to be genuine. The following error was caught:Failed to load library /home/user/ros2_mujoco_test/quadruped_ros2_control/install/hardware_unitree_mujoco/lib/libhardware_unitree_mujoco.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libddscxx.so.0: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
[rviz2-1] 1745044850.564918 [0]      rviz2: selected interface "lo" is not multicast-capable: disabling multicast
[spawner-4] 1745044850.576480 [0]    spawner: selected interface "lo" is not multicast-capable: disabling multicast
[spawner-4] [INFO] [1745044850.606871724] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[rviz2-1] [INFO] [1745044851.229644830] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1745044851.229695120] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1745044851.395813326] [rviz2]: Stereo is NOT SUPPORTED
[spawner-4] [WARN] [1745044860.617313195] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-4] [INFO] [1745044860.617613254] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...

我看到的主要错误是这些
[ros2_control_node-3] [ERROR] [1745044850.484965722] [controller_manager]: The published robot description file (urdf) seems not to be genuine. The following error was caught:Failed to load library /home/user/ros2_mujoco_test/quadruped_ros2_control/install/hardware_unitree_mujoco/lib/libhardware_unitree_mujoco.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libddscxx.so.0: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
[rviz2-1] 1745044850.564918 [0]      rviz2: selected interface "lo" is not multicast-capable: disabling multicast
[spawner-4] 1745044850.576480 [0]    spawner: selected interface "lo" is not multicast-capable: disabling multicast

但是libhardware_unitree_mujoco.so我的路径下是有的啊，还有就是前面fastdds其实没弄成功，换cunitree_ros2中的yclonedds配置了
setup_local.sh中写了这些
#!/bin/bash
echo "Setup unitree ros2 environment"
source /opt/ros/humble/setup.bash
source /home/user/unitreerobotics/unitree_ros2/cyclonedds_ws/install/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export CYCLONEDDS_URI='<CycloneDDS><Domain><General><Interfaces>
                            <NetworkInterface name="lo" priority="default" multicast="default" />
                        </Interfaces></General></Domain></CycloneDDS>'

legubiao, 2025-04-19:
感觉还是因为cyclonedds导致的。先把ros2 rmw换回默认的fastdds再尝试吧，必须要不用sudo就能打开unitree_mujoco，才能进行后续的步骤

illusoryTwin, 2025-05-08:
@lkn01 did you manage to resolve this issue? I also have this problem, I changed the dds type, but for me the problem remains

legubiao, 2025-05-12:
来自成功解决的网友：用 apt auto remove 彻底清理干净unitree ros2 和 unitree sdk2后， 重新安装

closed 2025-07-28 by legubiao