legubiao/quadruped_ros2_control #43: Unitree go2 spawns laying doing and no control from keyboard_input   翻译：Unitree go2 产生正在执行的操作，并且没有来自 keyboard_input 的控制

louisarbor, 2025-07-25:
Hello, 

Thank you for sharing your work!

I believe I have successfully built repository packages, but when launching Gazebo Harmonic, the unitree go2 spawns but in a laying down position (on its belly). 

I tried to use the keyboard_input, but it is unresponsive (I saw your response in this issue: https://github.com/legubiao/quadruped_ros2_control/issues/40 and I do select the terminal - but still no response). 

Questions:

1) Will the keyboard_input commands work if the robot spawns laying down?
2) If not, what command is required too get the robot standing up?
3) Is there anything else required to get the keyboard_input working and the robot moving in gazebo harmonic?

I would appreciate any information/clarification.

翻译：
您好，

感谢您分享您的成果！

我相信我已经成功构建了仓库包，但是启动 Gazebo Harmonic 时，unitree go2 生成了，但却处于平躺姿势（腹部朝下）。

我尝试使用 keyboard_input，但它没有响应（我看到了您在这个问题中的回复：https://github.com/legubiao/quadruped_ros2_control/issues/40，我也选择了终端，但仍然没有响应）。

问题：

1) 如果机器人生成时是平躺的，keyboard_input 命令还能正常工作吗？
2) 如果不能，需要什么命令才能让机器人站起来？
3) 还需要什么其他命令才能让 keyboard_input 正常工作，并使机器人在 Gazebo Harmonic 中移动？

如有任何信息/说明，我将不胜感激.

legubiao, 2025-07-25:
You can check the guide for `keyboard_input` at the readme file under keyboard_input folder.

louisarbor, 2025-07-25:
Thank you, 

There was some lag. I need to be more patient. :)

closed 2025-07-25 by louisarbor

YHLin117, 2025-09-03:
hi @louisarbor  have you solved the problem, i meet the same problem myself, the dog is in a laying down position, and the keyboard_input doesnt work

louisarbor, 2025-09-03:
Hello @YHLin117, 
I haven't worked on this for a while, but from memory as @legubiao mentioned, the instructions for keyboard_input are in the readme under the keyboard_input directory - follow these instructions (not the instructions in the terminal output). Also, control is slow, so be patient/hold down the keys long enough - and try different sequences. I remember I needed to follow specific sequences of keys to get it working. Ultimately keyboard_input is not what you want (control is not great, the robot falls over very quickly) - you'll likely want to implement some reinforcement learning control. :)

YHLin117, 2025-09-03:
@louisarbor THANK YOU SO MUCH, it works !